prusa3d/PrusaSlicer #9458: Feature Request: Automatically rename file uploaded via prusalink

cibomahto, 2023-01-31:
When exporting designs to my SL1S speed printer over network, I often get an error that the file already exists, and then the upload fails:

[image]

It would be great if the system could de-conflict this automatically, for example by appending numbers to the end of the name (-2, -3, etc) until it finds a filename that doesn't conflict, or by prepending the date and time to the name so that it would have a very low chance of conflicting. Ideally I wouldn't have to confirm anything.

For an example of this working well (though possibly unintentionally), files uploaded from PrusaSlicer to OctoPrint have the print time appended to the end of the file name, which also makes it unlikely that a job name will conflict- I haven't had it happen in a few years of usage:

[image]

kocikdav, 2023-02-01:
Hello.
You can customize the output filename in Print Settings -> Output Options. For example `{input_filename_base}_{printer_model}_{print_time}.sl1` might do what you are looking for.

closed 2023-02-01 by kocikdav

cibomahto, 2023-02-02:
Thanks! That's simple enough