prusa3d/PrusaSlicer #9459: Organic supports not supporting all overhangs, also no brim on bed

Arakon, 2023-01-31:
### Description of the bug

When auto-generating organic supports, several 90° overhangs are completely ignored. Using standard supports, they get supported.

Also, the support base is just 2 walls on the bed, using a brim around the support base similar to Cura's tree supports would greatly increase bed adhesion.

[image]
[image]

### Project file & How to reproduce

Project file is too large even when zipped, find it here:

https://drive.google.com/file/d/1bSJvaCZz40itTpNvdiPoDoq6fQwQ83KO/view?usp=sharing

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2

### Operating system

Windows 10

### Printer model

VZBot330

thunderpipesusa, 2023-01-31:
I cant even see how to turn on the organics.  i see settings for tree  but i cant choose between scaffold or tree

Arakon, 2023-02-03:
@ThomasRahm Exactly, that's near identical to how Cura's tree supports did it, and that print came out flawless.
Far as I can tell, everything that needs supporting, is supported now.

closed 2023-02-10 by bubnikv