prusa3d/PrusaSlicer #9461: View Rotates Continuously When Moving Mouse Without Clicking and Dragging

HarbingerOfYeet, 2023-01-31:
### Description of the bug

When I open up the slicer software, the bed view will continuously rotate whenever I move my mouse across the screen. I am unable to click the icon that allows for adding a new object. When I right click on the bed, the bed stops rotating until I left click. 

### Project file & How to reproduce

1. Open PrusaSlicer
2. Move mouse around without clicking and dragging and bed will rotate
3. Right click on bed to stop rotation
4. Left click away and bed will continue rotating with mouse movement

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0+linux-x64-GTK3

### Operating system

Pop!_OS 22.04 LTS (a version of Ubuntu)

### Printer model

Original Prusa i3

npike, 2023-02-10:
I also experience this, but on PrusaSlicer-2.6.0-alpha3+linux-x64-GTK3-202302031527

kubispe1, 2023-02-17:
At the moment we did not experience the same issue. We can not reproduce it.

wookieesnacks, 2023-03-20:
I am also experiencing in 2.6.0-alpha5+linux-x64-GTK3.

RunningUtes, 2023-04-27:
I am having the same issues. See [screencast video](https://www.dropbox.com/s/kzoqnhplx0bee9l/PrusaSlicer%20-%20viewport%20movement%20with%20mouse.mkv?dl=0). 
Version: 2.6.0-alpha6+linux-x64-GTK2
Build: PrusaSlicer-2.6.0-alpha6+linux-x64-GTK2-202303311518
Operating System: Unix
System Architecture: 64 bit

vx-github, 2023-10-08:
Experiencing the problem as described.

Version **2.6.1+linux-x64-GTK3** on Fedora

A reboot also resolved my problem.

vx-github, 2024-03-27:
After a while the problem popups again.

PrusaSlicer-2.7.2+linux-x64-GTK3-202402291307.AppImage

jbehrends, 2024-06-12:
PrusaSlicer-2.7.4+linux-x64-GTK3-202404050928.AppImage
Fedora Linux 39 (Cinnamon)
Lenovo Yoga 7i Laptop
#####################

This happens to me randomly all the time with both my laptop's touch pad, and a Bluetooth mouse.  This does not happen with any other application running on my system, just PrusaSlicer.

The only way I've been able to temporarily fix the issue to to unload/reload the "hid_multitouch" linux module using the following steps:

Open a Terminal
Then run:
```
sudo rmmod hid_multitouch
```
Then wait about 5 seconds, and or attempt to  move the mouse around.

Then run:
```
sudo modprobe hid_multitouch
```

After that the mouse in PrusaSlicer appears to work just fine for a while until it happens again..

brjhaverkamp, 2025-04-19:
I have the same problem poping up  regularly. Sometimes prusaslicer works correctly for days and then after coming back from a standby prusaslicer has this problem where I can't stop the buildplate from rotating with the mouse.

I tried the above trick, but removing and restarting hid_multitouch didn't help. I even removed all hid related modules and still the issue remained.
I am on a thinkpad 580, Ubuntu 24.04.2, Prusaslicer 2.9.2

Regards,

Bert

closed 2026-09-29 by prusaslicer-bot[bot]