prusa3d/PrusaSlicer #9462: Organic support with variable height

F-CJPJ, 2023-01-31:
### Description of the bug

I sliced with organic support : display is ok.

then I added the variable height , then I sliced again with the same support configuration.

The display shows that branches of the tree don't touch the ceiling of my part. See screen shot attached.

[Timeo_baton_v4.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/10550576/Timeo_baton_v4.gcode.zip)
[branchs_dont_touch](https://user-images.githubusercontent.com/95922940/215873265-64f89280-f4a0-464b-a95c-91a9fa6de077.png)

[Timeo_baton_v4.project.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10550606/Timeo_baton_v4.project.3mf.zip)

### Project file & How to reproduce

As explained above, the slicing with variable height shows that support branches don't touch the part !

With fixed 0.2 mm height, the branches are ok.

It might be a display issue or a slicing issue.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha2+MacOS-universal-202301311728.dmg

### Operating system

Mac0s Ventura 13.1 on MacMini M1

### Printer model

DIY (part not printed because printed time is 10 hours)

foreachthing, 2023-01-31:
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2
`Variable layer height is not supported.`

agravelot, 2023-01-31:
it can remain open as a feature request

kubispe1, 2023-02-20:
@agravelot, at the moment we let it in this stage. But it is on our list.

Lobuttomize, 2023-04-22:
Can someone explain why this is difficult to do? Not implying that it isn't, just want to know.

F-CJPJ, 2023-04-22:
Hello

In my case, when I mentioned the issue, it seemed to work BUT the top of 
the "trees" did not touch the part to print.

Now I tested again with alpha6 and the same part, but I can get variable 
layer height.

I don't understand.

TheBekker, 2023-05-11:
Just a +1 on this.
This would be amazing to have fixed.

muhmuhhum, 2023-06-09:
Also a +1 for me. Tried to print miniatures and want to printe the base in a lower resolution thant the rest to reduce printtime

foggyforests, 2023-06-19:
+1 for me as well. Variable layer height with snug supports takes a ton of material and time compared to organic supports. Would be wonderful to have fixed. Thanks for considering!

boodaah3d, 2023-06-26:
I understand the considerations that need to be taken for variable layer height with organic supports. But we should still be able to have variable layer height when those layers are above any of the printed supports.

freeformz, 2023-10-21:
+1 on what @boodaah3d mentioned. Should be easier to initially support variable layer height when the heights are above any organic supports.

VisualReversal, 2023-11-06:
I don't know if this is of any help but I came across this error today; along with a solution to my specific problem. In my case I was slicing a file that I made for the MK4 with a .4 nozzle and 0.30 layer height and 0.30 printer settings profile. However when I tried to transfer to an input shaper profile with 0.20, I got the error. Then I realized that I still had the 0.30 layer height in the range height modifier. Once I switched it to 0.20 to match the profile, the problem went away. Hope that helps.

Black6spdZ, 2024-03-11:
wouldn't it be simplest to make the organic support thickness follow the perimeter? I take it then infill combining also doesn't work with variable print height?

vgdh, 2024-03-31:
+1 I want this feature

jwidess, 2024-04-05:
I've also found that paint on supports get ignored when a height range modifier is used with organic supports. You can see what I'm talking about in the vid below.

https://github.com/prusa3d/PrusaSlicer/assets/24278368/03171b43-0ded-4795-95cb-30fa3d201b26

Tremetor, 2024-06-18:
+1 for organic supports with variable layer height.

Variable layer height is the best trade-off between speed and quality. And organic supports solve many use cases, where normal supports don't work. Combining both would be a great improvement!

HenkHavok, 2024-09-04:
+1 for organic supports with variable layer height.

This would make life a lot easier

FilmMartin, 2024-11-07:
Want to add my +1. Just printed a huge sphere. Variable layer height makes it possible to get a smooth top surface. Not being able to use organic support, meant the printing time increased from under 1 day to 2 ½ days -- and a _lot_ of wasted filament. Would LOVE the ability to combine variable layer height _and_ organic supports. Pretty please :)

Condyloma, 2024-11-30:
This feature is already available in some competitor slicers and offers a significant advantage by enabling faster prints while maintaining high detail in critical areas. Adjustable layer height is particularly important for older printers, as it allows for the best possible balance between speed and quality.

Drizzt321, 2025-03-10:
I'll add yet another "please add this" feature. I find myself often having to choose between organic supports that are sometimes/often easier to remove and adaptive height which can give me better vertically oriented curved surface accuracy and overall increased print speed.

e-basteln, 2025-04-20:
Another vote for this feature request. Organic support and variable layer height are a relevant combination for complex models. It seems that BambuStudio has this implemented, so could you just port the required changes back into PrusaSlicer?

mgebha, 2025-05-22:
Another vote from me: It is really sad for large prints to not be able to combine both features :-(

Blue-Crescent, 2025-06-07:
Thanks for great software. vote this feature

znsoft, 2025-11-30:
in cura  adaptive layer + tree support work fine

BastiBudde, 2026-02-20:
+1 for organic supports with variable layer height. This would really be a much-needed feature.

TGV1p3r, 2026-02-22:
+1 for organic supports with variable layer height, this feature would help a lot with large prints

zabelc, 2026-09-02:
This remains an issue in PrusaSlicer-3.0.0-alpha11 (on MacOS), which produces the following error message:

> Variable layer height is not supported with Organic supports.
See config keys: support_material_style

### Reproduction:
1. Import object with overhangs
2. Apply variable layer height (default parameters)
3. Set Generate Supports=On, Style=Organic
4. Slice -> Receive error message

kubispe1, 2026-09-07:
I converted this issue into a discussion. It makes sense to keep it. Thanks!