prusa3d/PrusaSlicer #9463: The "Open in PrusaSlicer" functionality on Printables doesn't work

DominikPalo, 2023-01-31:
### Description of the bug

The new "Open in PrusaSlicer" functionality on Printables doesn't work as expected. After clicking on the "Open in PrusaSlicer" button, the PrusaSlicet application is opened but no model is loaded (the workspace stays empty)

Example URL: `prusaslicer://open?file=https%3A%2F%2Ffiles.printables.com%2Fmedia%2Fprints%2F380760%2Fstls%2F3195166_b73bacc6-cc1b-4058-bc26-2b349b193229%2Fgeneral-pavel-flannel.stl`

### Project file & How to reproduce

1. Open any project on Printables.com
2. Go to the files tab
3. Click on the "Open in PrusaSlicer" near any STL file
4. PrusaSlicer app os properly opened, but the STL is not loaded

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2+MacOS-arm64

### Operating system

macOS 13.2

### Printer model

N/A

mix579, 2023-02-01:
Works for me. Same OS. Don't know why.

DominikPalo, 2023-02-01:
@Jan-Soustruznik thank you, it wasn't enabled in the slicer config (I didn't know about such an option). After enabling it, everything works as expected. Maybe it would be nice to display some warning message in case the app is triggered by a `prusaslicer://open?file=https%3A%2F%2Ffiles.printables.com...` URL and the setting mentioned above is disabled.

gudvinr, 2023-02-01:
> @Jan-Soustruznik thank you, it wasn't enabled in the slicer config (I didn't know about such an option). After enabling it, everything works as expected. Maybe it would be nice to display some [...]

printables.com says to enable this option when you enable checkbox in the menu

I am fairly certain that setting enables URI handler in first place, so without it app isn't triggered at all (I think)

DominikPalo, 2023-02-01:
> I am fairly certain that setting enables URI handler in first place, so without it app isn't triggered at all (I think)

That's not true - the custom `prusaslicer://` URL scheme is handled by the app even if the "Allow downloads from Printables.com" checkbox is disabled. The URL scheme is defined in the `Info.plist` file (`CFBundleURLTypes` key), so macOS associates the PrusaSlicer app with that scheme automatically (it is not even required to drag the app out from the DMG)

kocikdav, 2023-02-01:
To clarify: On both Windows and Linux it really is triggered only after the URL scheme is registered (so like @gudvinr wrote) but on Mac it is just as @DominikPalo writes. 
We will look into it so it won't happen exactly what happened here - PS pops up but nothing happens. 
Also Printables team was notified about the bug in translations.

closed 2026-09-29 by prusaslicer-bot[bot]