prusa3d/PrusaSlicer #9464: crash on slicing this file (at support material)

petrubecheru, 2023-01-31:
### Description of the bug

prusaslicer crashes when slicing the attached file, when generating grid supports. If I change to organic, it doesn't crash.

### Project file & How to reproduce

[Skeleton Hand Soap Dish - crash.zip](https://github.com/prusa3d/PrusaSlicer/files/10551649/Skeleton.Hand.Soap.Dish.-.crash.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha2

### Operating system

Ubuntu 22.10

### Printer model

Original Prusa i3 MK3S & MK3S+ MMU2S

lukasmatena, 2023-02-01:
Thanks, reproduced. It crashes with "Complete individual objects" on. It asserts on `assert(layer != nullptr);` in `GCode::sort_print_object_instances`. We will look into it.

bubnikv, 2023-02-03:
Fixed with 43dc3bfc429e65009369a43f705adb561293cdff

closed 2023-02-03 by bubnikv

lukasmatena, 2023-02-03:
Fixed in [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3).

commit 43dc3bfc (2023-02-03): Fix of a crash with sequential print and raft. Fixes #9464