prusa3d/PrusaSlicer #9465: All sliced files stop at firt layer, after the skirt.

flynetstudios, 2023-01-31:
### Description of the bug

Any file that i try to slice, just top the print in the first layer, after the skirt. This was testes with two different printers and with many files, i go to upload here two different gcodes, maybe you can find the issue here. I already reinstalled Prusa slicer 2.5 and deleted printers and presets, all reverted to stock.

### Project file & How to reproduce

[Uploading PrusaStop1stLayer.zip…]()

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0 x64

### Operating system

Windows 10

### Printer model

ender 3

lukasmatena, 2023-02-01:
I don't see any problem. The 3MF sliced fine in both 2.5.0 and 2.6.0, both G-Codes seem fine. Can you please check that the problem is not on your end? Thanks.

flynetstudios, 2023-02-02:
Did you try to print the gcode ?? The files slice perfect and i can see the simulation, but when i go to print the files, the print just stop and the printer get back to the ambient temp and the nozzle stay in the middle. Also, how can this be a problem on my end, if i can print other files, and also this is happening in two printers ?? Any suggestion about how to check it ??

lukasmatena, 2023-02-03:
@flynetstudios Well, at first I did not try to print it, I don't have an Ender. But I sliced the 3MF and checked the preview, checked your gcode in gcodeviever and inspected the first layer in notepad. I looked into the parameter tabs. Nothing caught my eye.

Now I tried to feed your G-codes into my MK3. Both printed fine for the first several layers, until I stopped them:
[image]

So I highly doubt there is a problem with the G-code.

> Also, how can this be a problem on my end, if i can print other files, and also this is happening in two printers ??

I don't know. We already saw mysterious issues and some of them had unexpected reasons. For example, corrupted file on a faulty SD card, firmware bugs, hw issues, etc.

flynetstudios, 2023-03-08:
After several test, i found the source of the issue. Im using beagle camera and ender 3 printer, with marlin 1.0.1. The problem is related to latest beagle camera firmware 1.7 and 1.6. Downgraded to firmware 1.5 and 1.4 and everything is working flawless. I already contaced to the Mintion team, and they are checking whats wrong with the firmware. Nothing wrong with Prusa Slicer after all. 

I hope this can help to other users facing the same issue.

lukasmatena, 2023-03-08:
Thanks for the update. Closing.

closed 2023-03-08 by lukasmatena