prusa3d/PrusaSlicer #9466: crash on generating perimeters

petrubecheru, 2023-01-31:
### Description of the bug

prusaslicer crashes of generating perimeters of the sheep model (MMU).

### Project file & How to reproduce

[oaia-crash.zip](https://github.com/prusa3d/PrusaSlicer/files/10551772/oaia-crash.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha2

### Operating system

Ubuntu 22.10

### Printer model

Original Prusa i3 MK3S & MK3S+ MMU2S

lukasmatena, 2023-02-01:
Thanks, reproduced. It crashes on `std::swap(layerm.m_fill_expolygons[region_fill_id], layerm.m_fill_expolygons[last]);` in Layer.cpp while attempting to swap two elements of an empty vector. We will fix this.

bubnikv, 2023-02-02:
Fixed with 3cb2f5f58f952bc104260c84292411894f957552

kubispe1, 2023-02-03:
As [@bubnikv](https://github.com/bubnikv) mentioned issue was fixed and tested.
I am closing
[image]

commit 3cb2f5f5 (2023-02-03): Fix of Layer::sort_perimeters_into_islands()
Fixes #9466 #9511
closed 2023-02-03 by kubispe1