prusa3d/PrusaSlicer #9468: Allow the Seams to be visible at all times

thunderpipesusa, 2023-01-31:
Please make the seams option as an always on or off toggle.  maybe in the preferences under gui? its super annoying when im in a rush and forget to show the seams, then find out too late after a print has started that i forgot to view them lol.  Would just be nice if seams was on by default

igiannakas, 2023-02-01:
Agree, that would be nice!!

foreachthing, 2023-02-01:
It should "just" remember last used options. Done.

mix579, 2023-02-15:
This is such a nuisance indeed! Remembering what options you've chosen feels like a no-brainer.

At least with 2.6 I don't have to click TWICE on the drop down to select seams (or whatever option) anymore, now that they have those (indecipherable) icons.

revolter, 2023-04-02:
I also agree with this. Even for newbies that aren't yet interested in this kind of details, or even for newbies that don't even know what seams are, having them enabled:
- isn't disrupting or confusing, and
- is intuitive enough to make them search and learn about them.

And, obviously, the OP's issue, which is exactly mine: forgetting to check the seams before exporting, or even worse, before printing.

As such, I created a PR to address this.

revolter, 2023-04-11:
Aaaand, another suboptimal print because I forgot about the seam 😞

thunderpipesusa, 2023-05-18:
is this being worked on?

revolter, 2023-05-19:
A member needs to review my PR.

RickX56, 2023-05-28:
That would be great having the seams showing on by default when sliced or at least an option to always show in User preferences. So many great prints would be saved!

xi2023, 2023-06-16:
PrusaSlicer 2.6.0-rc2 has been ''pre-released'' 2 days ago and it looks like there has been no changes at all... i mean... how long would it take to fix this for someone that knows what he is doing? 5 or 10minutes? we are asking for this for months now, multiple times and it is just getting ignored... sorry but at this point its getting ridiculous

revolter, 2023-06-17:
> PrusaSlicer 2.6.0-rc2 has been ''pre-released'' 2 days ago and it looks like there has been no changes at all... i mean... how long would it take to fix this for someone that knows what he is doing? [...]

Again, it doesn't need to be fixed, because I already did that, but, for some reason, they won't merge this simple change.

xi2023, 2023-06-17:
have you tested it to see if it works? for me it would be fine if the seam is always visible but new users might start asking what the white dots are. this would be still better then people asking why the print looks ugly. 
anyway, just to ''remember'' what we have selected or a option in the preference-GUI tab to always show the seam would be possibly the better soulution for everyone. but that would probably need more than 2 lines.
changing a few lines would be not that hard, many of us have already done this with firmware, but building/compile a slicer seems more confusing, i did take a quick look at it and no thanks... i rather change to a different slicer.

Tupson444, 2023-06-17:
>... this would be still better then people asking why the print looks ugly.

This is a good point - it would be a good thing if someone wonders "what is this" in the slicer, instead on a finished print.

But there could be a tickbox in the general settings or something, so those who want it can turn on "seams always visible" without bothering those who don't want it.

mix579, 2023-06-17:
I think the best solution would be a checkbox under Preferences to "remember Preview settings" so it's not just seams but whatever people want to see.

xi2023, 2023-06-21:
PrusaSlicer 2.6.0 (20.6.2023) has now been officially released, it was OFFICIALLY released with BUGS and NOT what we asked for multiple times! at this point i am DONE with PRUSA! for me, its just ridiculous, to add USELESS featuers and not something just so simple, but usefull, as just remember/save the option to show the seams. it would save you from F* up prints and will give you controll over something you will not notice when you in a hurry or don't think about it.... im 3d printing for a few years now,  i have (just) three 3dprinters running and i can only advise new users not to use prusa slicer, just because of this one option. it is sad but true, a simple option can cost you several prints, hours worth of wasted time and filament and it's just not worth it! there a other good slicers out there, that will you give the same results and not waste your pressureses time! enough said, im just done with prusa slicer at this point , i will keep track of the development but i dont have much hope that the people in charge will change something about it anytime soon, is has been too long that we asked and nothing happend. good bye prusa

catequalsgood, 2023-07-30:
This would probably be my number one wish for a new feature. It is SO disappointing to take a good part off of the printer just to discover, that the seam was in the worst place because you did not check

revolter, 2023-12-07:
> I think the best solution would be a checkbox under Preferences to "remember Preview settings" so it's not just seams but whatever people want to see.

I would argue that it would make even more sense to NOT have a checkbox under Preferences, but rather have the preview settings states automatically persisted between sessions.

Why would I want to have the app behave like it's the first time I open it after installation?

Thyraz, 2024-05-01:
Please, please make the application remember the last view settings here.
I forgot about the seam setting so often and hate the result after a long print time ...

Trini8ed, 2025-04-29:
Prusa please do this for, don't do it for me or anyone else. Do it for the sake of man kind.

caste1200, 2025-11-30:
how is this not been implemented yet?? this is why I love orca slicer :(

caste1200, 2025-11-30:
> I'm curious how and why issues like this get completely ignored. It's an open source project, so if someone wants something they can contribute by writing code for it - which is what happened here, it just needs to be added. How can such a simple fix that would notably improve experience of users be such a low priority?

I wish I knew how to do it, i'd do it myself haha

MOtt, 2025-12-14:
Again a print for the bin  (okay, my fault - i've forgot to check the seams...)

Please activate "Show seams" by default or store the selected value as a session setting

vivienm, 2026-09-02:
Nice try, Prusa Bot, but we still want this feature ;)

gabrielkiss, 2026-09-02:
@vivienm maybe the new community plugins will offer a solution to this?

mix579, 2026-09-02:
I've also create a new request at https://github.com/prusa3d/PrusaSlicer/discussions/15616

SachCZ, 2026-09-18:
Please go show the support to the feature request here: https://github.com/prusa3d/PrusaSlicer/discussions/15616 it already has a lot of upvotes and is in our field of view. It is the best way to show us that you really care. Thank you, I am closing this issue, as from now only reproducible bugs belong to issues.

closed 2026-09-18 by SachCZ