prusa3d/PrusaSlicer #9469: Zombie Profile Import

bobbyh41, 2023-01-31:
### Description of the bug

This is not a new problem, I have it every time I install a alpha, beta, and even real releases on my M1 Imac. I have old profiles that I have deleted long ago and every time i load a new version like the new 2.6 beta i just installed, it will ask me to import from my previous version, I say yes, and not only does it import the correct profiles that I currently use in the release version, it also imports many old deleted profile that have been gone for a while, then I have to go threw and delete them all.

### Project file & How to reproduce

Just with the built in import on first launch on my M1 Imac.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6 Alpha 2

### Operating system

MacOS 13.1

### Printer model

Mk3S+

kocikdav, 2023-02-01:
Hello, thank you for reporting. Would you help me with these questions:
What kind of profiles are these old profiles? A custom printer or customized system profiles or someting else? And how do you delete them? Inside PrusaSlicer or by deleting files in Finder?

bobbyh41, 2023-02-01:
The only things i am noticing come back are both under printer: and its both user presets and physical printers. I delete them by going to the printer settings tab, selecting the printer and clicking the red X.

neophyl, 2023-02-01:
Were they previously deleted from the alpha directory/install before you tried the 2.6 alpha ?  In other words were they left over from the last time you tried an alpha.

bobbyh41, 2023-02-01:
All I do when I delete an alpha/beta is trash the icon, I don’t actually know where any other files are. But when I loaded the alpha, it asked about importing from my current 2.5 install. But it still pulled old profiles that are not in my current 2.5.

neophyl, 2023-02-01:
That probably explains it then.  Alphas, betas and release all store to different locations, so that you can run an early release and not effect your 'working' versions.  Removing the program or the icon does not remove those profile folders.  

So if you had an alpha on before then when opening 2.6 it would load in the original alpha profiles from the last alpha you had and then ask you about importing the ones from your current release version, so in this case 2.5.  You would then have the original alpha profiles and your current 2.5 ones stored into the alpha location.  

Same with a beta once 2.6 gets to that stage.

bobbyh41, 2023-02-01:
Ok, I guess I need to figure out where on the Mac the old profiles are hidden. But I do feel like if I delete things using the program, it should delete them so they don’t come back. But I guess it’s only a problem with people who use the alphas and betas, so it’s probably not that big of a deal.

kocikdav, 2023-02-01:
@bobbyh41 It could be its not really hidden anywhere. The case could be:
You had the old profiles in 2.5.0-alpha -> Then you imported it to release version 2.5.0 -> you deleted the old profiles -> you imported your configuration from 2.5.0 to 2.6.0-alpha. Which means you've got your last configuration into folder where your old profiles still are. Thus you had to delete it again.

Please note that there is different config folder for alpha, beta and release versions. So this might happen once more when you import from alpha to beta.

I understand this is inconvenient for you, but it is expected behavior so im closing this issue.

closed 2023-02-01 by kocikdav

lukasmatena, 2023-02-02:
@bobbyh41
> Ok, I guess I need to figure out where on the Mac the old profiles are hidden.

`Help->Show Configuration Folder` will show you current config directory. You'll notice that it is indeed different for alphas/betas/finals.

bobbyh41, 2023-02-02:
Ok, thanks for the help, and for others having this problem, on the mac you have to figure out how to unhide hidden files and folders to find it, took me some time. I'm still a little confused as to why when it loaded I told it to import from my version of 2.5 but instead it pulled the profiles from the old beta folder, it still does not seem like proper functionality.

neophyl, 2023-02-02:
It loads in the old alpha profiles  as that's what it has when it starts AND then it offers you the option to import the profiles from your active 'release'.  Its normal and correct operation.  If you choose not to import it would have just started with the ones from your old alpha install.