prusa3d/PrusaSlicer #9470: Program crashes on slicing

terryrobison, 2023-02-01:
### Description of the bug

I loaded the official benchy STL model.  I then rotated the model by 45 degrees.  Selected Organic Supports, then slice.   Program crashes, are repeatable.

I was using tool 2 on my E3D Toolchanger profile.

### Project file & How to reproduce

I loaded the official benchy STL model.  I then rotated the model by 45 degrees.  Selected Organic Supports, then slice.   Program crashes, are repeatable.

I was using tool 2 on my E3D Toolchanger profile.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha2+win64

### Operating system

Windows 10

### Printer model

Offical Benchy Model

lukasmatena, 2023-02-01:
Please, provide a 3MF. Thanks.

mann1861, 2023-02-01:
I have the same issue with multiple extruder. Slices fine on tool 0 but crashes during slicing on others. Here's a 3mf with a benchy that showed this behaviour
[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/10555237/3DBenchy.zip)

lukasmatena, 2023-02-01:
@mann1861 Thanks, but I cannot reproduce it. Your 3MF slices fine on my system in 2.5.0 and 2.6.0-alpha2, regardless of the selected extruder. Can you please check that the 3MF indeed captures the crashing scenario or possibly include a short video showing all the steps between starting the application and the crash? Thanks.

terryrobison, 2023-02-02:
Choosing toolhead 1, then selecting toolhead 2.  Crashes on T1 but not T0.   Slicing organic supports.

[https://www.dropbox.com/s/wf8eo2uokmgq929/_Untitled%20-%20PrusaSlicer-2.6.0-alpha2%20based%20on%20Slic3r%202023-02-02%2014-40-19.mp4?dl=0
](url)

[https://www.dropbox.com/s/uybkb8or17yjoqf/_Untitled%20-%20PrusaSlicer-2.6.0-alpha2%20based%20on%20Slic3r%202023-02-02%2014-56-38.mp4?dl=0](url)

lukasmatena, 2023-02-03:
@terryrobison Thanks, but still unable to reproduce it. I need the 3MF which I asked for.

terryrobison, 2023-02-05:
[files.zip](https://github.com/prusa3d/PrusaSlicer/files/10610454/files.zip)

kubispe1, 2023-02-06:
I took one of your previous 3mf and have followed your videos. The issue is reproducible with direct crash or error message Acess violation...
I think this issue is duplicite of #9550 
I am closing it. 
[3DBenchy_crash.zip](https://github.com/prusa3d/PrusaSlicer/files/10661639/3DBenchy_crash.zip)
[image]

closed 2023-02-06 by kubispe1