prusa3d/PrusaSlicer #9471: Cut tool suggestion

michaudtime, 2023-02-01:
I love that the cut tool is getting some love! One of the things is that its a little hard to find the exact point you want to make the cut on.

**Describe the solution you'd like**
What I've found that ive been doing is placing the cut line and then clicking add/edit connectors as it hides the top and and I can see whats left. If this could be added as a toggle in the normal cut tool it would be a big QoL improvement. This allows you to easily see if there might be a  random island that you overlooked initially. 

Great work as always team!

lukasmatena, 2023-02-01:
> This allows you to easily see if there might be a random island that you overlooked initially.

I am not sure I understand. This is the reason why we are showing the cut contour. Is that not enough?

michaudtime, 2023-02-01:
While the contour works as expected it would just be a little easier to see when your messing up. I've attached an very primitive example.  I would like to split this part in 2 parts to print as flat on the bed as possible so can maximize overall surface finish. In photo 1 I have put the cut tool where I thought it would work best. After pressing the connector button (photo 2) You can easily see that I missed a surface that would potential drive the need for supports which I don't want to use. I would then hit cancel and readjust the cut line until I had something to match photo 3. 

It's not a huge deal, just a small QoL thing. 

[image]
[image]
[image]

closed 2026-09-29 by prusaslicer-bot[bot]