prusa3d/PrusaSlicer #9473: Place on cut toggle does not work

gudvinr, 2023-02-01:
### Description of the bug

When you disable "Place on cut" for the top part, it still gets placed on cut (i.e. behaviour does not change).

### Project file & How to reproduce

1. Open cut panel
2. Toggle "Place on cut" off for top part
3. Perform cut

[Full.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10552354/Full.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2+linux-x64-GTK3

### Operating system

Arch Linux

### Printer model

Artillery Genius Pro

lukasmatena, 2023-02-01:
I assume that you are cutting horizontally. In that case, there is indeed no difference between placing the top part on cut and leaving it as it is. Tilting the plane however makes the two outcomes different.

If that's not what you meant, please provide some pictures.

gudvinr, 2023-02-01:
No, what I mean is that it is very contradictory to what actually happens.

What I am expecting (and I am not saying that I am right), that if I remove "Place on cut" on the "Upper" part, I'd expect pieces just be cut without any translation.

Why do I think that, you might ask? Okay, let me explain what happens _now_.
First, on the lower part "Place on cut" is not pressed, and it doesn't move. So naturally I think that when you do the same on "Upper" part, same thing will happen. That could be just personal preference, so I am not being picky about this.

But in reality, there are no "Upper" and "Lower" parts, there is a part that on the side of the plane where normal is and one that is on the opposite side.  
When you rotate plane 180°, what you call "Upper" part, will be actually be closer to buildplate and if you do a cut this way, that'll yield technically correct but very confusing result because _actual_ upper part will be at the bottom.

closed 2026-09-29 by prusaslicer-bot[bot]