prusa3d/PrusaSlicer #9476: PrusaSlicer2.6alpha2 Switching PrusaSlicer to language zh_TW failed.

wisonlau, 2023-02-01:
### Description of the bug

[image]

### Project file & How to reproduce

Open the slicing software and it will appear.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha2+MacOS-universal-202301311728.dmg

### Operating system

macos 12.6.1

### Printer model

prusa mk3s

lukasmatena, 2023-02-02:
@wisonlau Do you have the same problem with PrusaSlicer 2.5.0, or is it something new in 2.6.0-alpha2? If the latter, it could be a consequence of the wxWidgets upgrade.

wisonlau, 2023-02-02:
> 

> @wisonlau Do you have the same problem with PrusaSlicer 2.5.0, or is it something new in 2.6.0-alpha2? If the latter, it could be a consequence of the wxWidgets upgrade.

PrusaSlicer 2.5.0 can be opened normally,It should be the problem of version 2.6.0

lukasmatena, 2023-02-02:
> What should I do?

I don't know. We will investigate and let you know. Thanks for reporting it.

wisonlau, 2023-02-02:
> > What should I do?
> 
> I don't know. We will investigate and let you know. Thanks for reporting it.

When an error prompt pops up, can the startup process log be displayed for troubleshooting

xihrni, 2023-02-03:
I had the same problem with an apple chip mac, macOS 13.1

lukasmatena, 2023-02-03:
> When an error prompt pops up, can the startup process log be displayed for troubleshooting

You can run the application from the command line with a parameter: `./prusa-slicer --loglevel=5` to enable logging to stdout and them post the output here. I don't think it will help much in this case, but if you do it, we will check. Thanks.

wisonlau, 2023-02-03:
> You can run the application from the command line with a parameter: `./prusa-slicer --loglevel=5` to enable logging to stdout and them post the output here. I don't think it will help much in this [...]

```
(base) ➜  MacOS ./PrusaSlicer --loglevel=5
[2023-02-03 19:37:35.311688] [0x0000000118845600] [trace]   Initializing StaticPrintConfigs
[2023-02-03 19:37:35.402911] [0x0000000118845600] [trace]   App updater default download path: "/Users/wisonlau/Downloads"
[2023-02-03 19:37:35.403055] [0x0000000118845600] [debug]   full path: 4072512567279585534
[2023-02-03 19:37:35.403080] [0x0000000118845600] [debug]   single instance: undefined. other params: ./PrusaSlicer;--loglevel=5
[2023-02-03 19:37:35.403101] [0x0000000118845600] [debug]   full lock path: /Users/wisonlau/Library/Application Support/PrusaSlicer-alpha/cache/4072512567279585534.lock
[2023-02-03 19:37:35.403178] [0x0000000118845600] [debug]   Creating lockfile.
[2023-02-03 19:37:35.403186] [0x0000000118845600] [info]    Instance check: Another instance not found or single-instance not set.
[2023-02-03 19:37:35.525108] [0x0000000118845600] [trace]   System language detected (user locales and such): zh_CN
[2023-02-03 19:37:35.530705] [0x0000000118845600] [trace]   Best translation language detected (may be different from user locales): zh_TW
[2023-02-03 19:37:35.530956] [0x0000000118845600] [trace]   Switching wxLocales to zh_TW
```

This is the operation log

lukasmatena, 2023-02-03:
@wisonlau Thanks. We managed to reproduce the issue, we will try to fix it. In the meantime, try the freshly released [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3). We at least bypassed the problem so the application will run. You will see an error message at the start and it will run with English locales. Better than nothing for now.

wisonlau, 2023-02-04:
> @wisonlau Thanks. We managed to reproduce the issue, we will try to fix it. In the meantime, try the freshly released [...]

Thank you. This is feasible. I look forward to the new version.

kubispe1, 2023-02-16:
fixed with commit: 1cfa5ee38522aa2fc5f2d287c67cd966a9a88aea
SPE-1457
Closing this issue. Thanks

closed 2023-02-16 by kubispe1