prusa3d/PrusaSlicer #9477: Organic Supports Will not Turn off

CaptainCodswallop, 2023-02-01:
### Description of the bug

I loaded a model and used organic supports. Afterwards I attempted to turn off the organics supports by selecting "no supports". The supports are still being generated.

### Project file & How to reproduce

[Calibration Cube.zip](https://github.com/prusa3d/PrusaSlicer/files/10552791/Calibration.Cube.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusa Slicer 2.6 alpha 2

### Operating system

Windows 11

### Printer model

Prusa Mini

lukasmatena, 2023-02-01:
Thanks. It seems to be caused by "Print Settings->Support Material->Enforce supports for the first N layers". It appears to work with Grid and Snug supports, but with Organic it turns into an on/off switch.

CaptainCodswallop, 2023-02-01:
You are welcome. Been very much enjoying the new features. Hope easy fix.
Cheers

bubnikv, 2023-02-08:
fixed with a1b0188e3a04270b18455404e6a79cc130df49da

closed 2023-02-08 by bubnikv