prusa3d/PrusaSlicer #9479: PrusaSlice stop Work

marcelodumke, 2023-02-01:
### Description of the bug

I sliced ​​the project below, and the travel time is too long, when I send it to show the travel time, the program is extremely slow. see the images, I'm forwarding the project together

### Project file & How to reproduce

Open project and slice, in my creality ender 3 S1 pro
[image]
[image]
[simpsons001.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10553240/simpsons001.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha 2 MacOS

### Operating system

MacOS 13.2

### Printer model

Creality ender 3 s1 pro

kubispe1, 2023-02-07:
I think it is affected by your speed settings of **Z travel speed**. When you put there something more appropriate (let's say 12mm/s) or default speed ( switch by input 0) it reduces travel time to 1h35m up to 2h.

lukasmatena, 2023-02-07:
The UI lag when travels are shown is also reported in #9635.

lukasmatena, 2024-02-19:
This should be already fixed. Closing.

closed 2024-02-19 by lukasmatena