prusa3d/PrusaSlicer #9480: extra perimeter on overhangs starts in the air

SphaeroX, 2023-02-01:
### Description of the bug

The print starts here in the air, if it would start properly at the print part then one could print very well in the air like for example with ArcOverhangs

[image]

### Project file & How to reproduce

[09_support.zip](https://github.com/prusa3d/PrusaSlicer/files/10553640/09_support.zip)

Open with 2.6 Alpha

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha2+win64

### Operating system

Win 11

### Printer model

custom

kubispe1, 2023-02-02:
Thanks for your issue. We are working on fix.

Godrak, 2023-02-16:
We have fixed the issue where the extra perimeter connected on a wrong spot to the previous perimeter and then the pattern broke. This could happen especially when there is a  single extrusion line gap fill as last extrusion, as in your example. The extrusions should also not break in the air as happened in your example. 

However, as for the extrusions starting in the air, this can still happen, and it is usually signal that you should enable supports. The extra perimeters feature is not supposed to allow printing of big overhangs without supports. It reduces the amount of supports needed (with supports autopaint tool), tries to avoid unanchored bridges in the air and improves the surface quality.

closed 2023-02-16 by Godrak

dsovsyannikov, 2023-03-09:
[image]

dsovsyannikov, 2023-03-09:
[image]

dsovsyannikov, 2023-03-09:
[kp5l v1.zip](https://github.com/prusa3d/PrusaSlicer/files/10935205/kp5l.v1.zip)