prusa3d/PrusaSlicer #9481: ironing doesn't work when no top layers

codecranker, 2023-02-01:
### Description of the bug

I'm printing with 100% dense infill and no top or bottom layers. I know it's a dumb setting, but ironing should still work. 

### Project file & How to reproduce

Turn on ironing all top surfaces. Set top and bottom layers to zero. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2-win64

### Operating system

Windows 10

### Printer model

trashheap1.0

kubispe1, 2023-02-01:
I don't understand what is the purpose of your issue. But that's true that when you switch top layers, there will not be applied ironing.

sarusani, 2023-02-01:
The documentation is correct. It says it will iron the top layer. You turned off top layers, so there's nothing to iron. If you want ironing use at least one top layer.
The same logic applies to top layer infill pattern. If you turn off top layers it will be ignored and not applied to the top infill instead. It's consistent.

codecranker, 2023-02-03:
This is annoying but I'll live with it and spare you the GitHub issue

closed 2023-02-03 by codecranker