prusa3d/PrusaSlicer #9482: Problem with Support material if using Variable layer height function

rainell, 2023-02-01:
### Description of the bug

When using Variable layer height function the Support material goes crazy. The problem is verified with many models (STLs), but not with all models. In the first attached image the Variable layer height function is not used, in the second image it is used.
[image]
[image]

### Project file & How to reproduce

Just use Variable layer height function
[Hulk.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10553892/Hulk.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

 PrusaSlicer-2.6.0-alpha2+MacOS-universal-202301311728.dmg 

### Operating system

macOS 10.15.7

### Printer model

Any printer (MK3S, Artillery X1, Ender 3)

neophyl, 2023-02-01:
Please read the release notes for 2.6, Variable Layer Height and and organic supports are not compatible.

The Organic supports currently pose the following limitations:

For large models with a lot of supports, 16GB RAM minimum is recommended.
Top / bottom vertical separation distances are rounded to multiples of layer height.
**Variable layer height is not supported.**
Interface layers are not always produced reliably. As of now printing with zero interface layers is recommended, which produces easy to remove supports.
Rafts with Organic supports are buggy.

rainell, 2023-02-01:
Thanks for the info

closed 2023-02-01 by rainell