prusa3d/PrusaSlicer #9483: X-Y-Z lines when zooming

jiiimi, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
When zooming, the lines (red-blue-green) of the X-Y-Z axes always have the same diameter.
Therefore a 1mm diameter hole is blocked by the blue Z axis and therefore we no longer see the inside of the hole.

**Describe the solution you'd like**
When you zoom in, the centerlines should decrease in diameter to allow a good visualization of the inside of the small holes.

[image]

stepikovo, 2023-03-02:
Can you please describe to me how you managed to get this view?

jiiimi, 2023-03-03:
There is a hole in the center which is diameter 1.00
You can't see the inside of this hole because the Z axis blue line is too thick.
When zooming in or out, the lines of the X Y Z axes always have the same thickness.

neophyl, 2023-03-03:
Jiiimi, the rerason you were asked is thats not a 'normal' view at all.  When sliced there are usually no coloured lines at all, for example
[image]

Those colour axis indicatorts usually are only shown on the no slice preview, they are also usually VERY much thinner.  
[image]

So we are wondering exactly how you get them to view like you have pictured ?

enricoturri1966, 2023-03-03:
@neophyl 
You can get to the reported state by changing the origin of the printbed.
The shown arrows are the reference system axes placed at that origin.

neophyl, 2023-03-03:
Ah cool, so like a delta setup then ?  I'll have to give that a go.

jiiimi, 2023-03-03:
when I open a part, 3d editor view, the X-Y-Z centerlines are visible and too thick:
[image]

after cutting, preview view, X-Y-Z centerlines are visible and too thick:
[image]

lukasmatena, 2023-04-02:
Automatic downscale of the reference axes was implemented in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6). Closing.

closed 2023-04-02 by lukasmatena