prusa3d/PrusaSlicer #9485: Dynamic overhang speed does not respect pressure equalizer limits

murk-sy, 2023-02-01:
### Description of the bug

When enabling both Dynamic overhang speed and Pressure equalizer, the max volumetric slope is not respected when entering the dynamic overhangs. Both settings are experimental.
Side note: Dynamic overhang speed is not listed as experimental in the slicer, while Pressure equalizer is.

[image]

[image]

### Project file & How to reproduce

1) Insert model with overhangs (45° is fine)
2) Enable both Dynamic Overhang Speed and Pressure equalizer
3) Set both max volumetric slopes to a low number (such as 0,5 mm^3/s^2) for best visibility

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2

### Operating system

Windows 10

### Printer model

Prusa Mini

kubispe1, 2023-02-09:
Thank you for your cooperation issue was tested. The fix will be part of the next release.

igiannakas, 2023-02-19:
Issue appears unresolved in the latest alpha. I still get abrupt speed change on overhang perimeters despite the volumetric slope being enabled.

igiannakas, 2023-02-19:
This pull request fully fixes it. I have manually merged it on my local machine and compiled it and the problem is solved:
https://github.com/prusa3d/PrusaSlicer/pull/9622

kubispe1, 2023-03-08:
@murk-sy can you retest with new 2.6.0-alpha5. There are  fixes to this issue.
> Fixed "Dynamic overhang speed" when Pressure equalizer was used (Changelog)

Thanks

murk-sy, 2023-03-09:
@kubispe1 Does not appear to fix the issue

Same model:
[image]

Sample project:
[image]

[2023-03-09 dynamic overhang speed - pressure equalizer.zip](https://github.com/prusa3d/PrusaSlicer/files/10928744/2023-03-09.dynamic.overhang.speed.-.pressure.equalizer.zip)

zxy16305, 2023-09-11:
in 2.6.1 has same problem, change pressure equalizer value not effect overhangs speed.

pressure equalizer set 10:
[image]

 pressure equalizer set 0.1:
[image]

Jaze20, 2023-12-27:
As of 20231226 and PrusaSlicer 2.7.1.  This issue still exist and causes very noticeable defects when dynamic overhang is at 5mm/s to fix curling on petg while the perimeter print speed is 40+mm/s  Also the acceleration and decceleration happens way to early slowing down the whole print.  Very annoying

igiannakas, 2023-12-27:
> As of 20231226 and PrusaSlicer 2.7.1. This issue still exist and causes very noticeable defects when dynamic overhang is at 5mm/s to fix curling on petg while the perimeter print speed is 40+mm/s [...]

Your best bet is to use orca slicer to be honest. I have ported over the pressure equalizer feature (extrusion rate smoothing) and have implemented the fixes proposed a year ago to respect pressure equalizer limits on external perimeters. Don’t think prusa will address this as, from what I see, don’t see the need for pressure equalizer on external perimeters at all.

igiannakas, 2023-12-27:
[image]

Jaze20, 2023-12-27:
OK thank you, I'll give orca a shot.  Did not want to relearn/port over all
my settings.  Looks like I have no choice.  Why do you think prusa don't
see a need for pressure equalizer on external perimeters?  Is it because
they print slowly?

closed 2024-03-21 by lukasmatena