prusa3d/PrusaSlicer #9486: fix print in the air

dsovsyannikov, 2023-02-01:
on some models, the beginning or end of the line is not correct.
line starts or ends in the air.

I propose a solution to this problem.

on overhangs
change the order of the lines. first the lines that have the maximum contact with the already printed lines.
the beginning and end of the line must touch the already printed line

murk-sy, 2023-02-01:
Can you provide example screenshots and project?

dsovsyannikov, 2023-02-01:
[image]

dsovsyannikov, 2023-02-01:
[image]

dsovsyannikov, 2023-02-01:
[disk.zip](https://github.com/prusa3d/PrusaSlicer/files/10555862/disk.zip)

murk-sy, 2023-02-01:
Try the new 2.6.0-alpha2 with "Extra perimeters on overhangs" setting.
It is not a 100% solution, but it is an improvement in some cases. Providing examples of where it doesn't work (but seems like it should) will also be useful for developers.

[image]
[image]

dsovsyannikov, 2023-02-01:
the conditions I wrote will greatly improve the printing of overhangs and bridges

I hope the developers will accept this.

Hyperspeed1313, 2023-02-01:
[image]
Managed to find another edge case with this Pokémon figurine on a stand turned upside down (had to enable some amount of support to get it to use Extra Perimeters on Overhangs at all). The image is of the first layer of the figurine's base.

The algorithm is attempting to draw using the target perimeter of the current layer, which works when the expansion isn't that much, but the algorithm should utilize the previous layer's geometry until it can draw an entire perimeter that's touching adjacent lines using the current layer's perimeter shape. In this model's case, that would be expanding out from the feet and moving inwards, then drawing out from the center in the new layer's perimeter shape.
[Upside-down Lucario Test.zip](https://github.com/prusa3d/PrusaSlicer/files/10560082/Upside-down.Lucario.Test.zip)

kubispe1, 2023-02-24:
there some comment to this issue. https://github.com/prusa3d/PrusaSlicer/issues/9480#issuecomment-1433058152

dsovsyannikov, 2023-03-09:
[image]

dsovsyannikov, 2023-03-09:
[kp5l v1.zip](https://github.com/prusa3d/PrusaSlicer/files/10935222/kp5l.v1.zip)

Godrak, 2023-03-10:
It will be fixed in the next release. The algorithm will choose to print anchored extrusions first. Note that this on its own does not ensure printability, as on complex overhangs the extrusion may be too long or turn extensively, and two anchored endpoints will make little difference there. However, the new version will try to make the path ordering much more sensible.

closed 2023-03-10 by Godrak