prusa3d/PrusaSlicer #9487: Organic supports misplacement in combination with adaptive variable layer height.

matousek35, 2023-02-01:
### Description of the bug

Organic supports are misplaced when adaptive variable layer height is used.

### Project file & How to reproduce

[image]
[image]
[projects.zip](https://github.com/prusa3d/PrusaSlicer/files/10555720/projects.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha2

### Operating system

Windows 10

### Printer model

MK3S+

K1-K1, 2023-02-01:
I can confirm this
[image]

Eldenroot, 2023-02-01:
Check changelog, I think it is not possible to use it with adaptive layer height.

neophyl, 2023-02-01:
I do wish people would actually read the release notes lol.  I think this is the second one of these today.

matousek35, 2023-02-01:
Sorry, I really should have read the changelog.

closed 2023-02-01 by matousek35