prusa3d/PrusaSlicer #9489: Access Violation Error in PrusaSlicer 2.6.0 Alpha 2

jdemacek, 2023-02-01:
### Description of the bug

When slicing a complex model with Multi Color parts, I receive an access violation error.  No other data given other than a recommendation to restart.  I an including a screenshot as well as my project file.

Project File:
[https://drive.google.com/file/d/1vRaV3LDxpyj1WNHy9ZCMY2DuwUStjDf4/view?usp=share_link](url)

Screenshot:
[https://drive.google.com/file/d/1PDYkHbq-wYGNW-G7ct7_yz_7Nr2rsP5O/view?usp=share_link](url)

### Project file & How to reproduce

Load linked file and "slice"

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0 Alpha 2

### Operating system

Windows 11 64 bit

### Printer model

Prusa MK3S / MMU2S

lukasmatena, 2023-02-01:
Thanks, reproduced. I simplified the project to run faster in debug mode: 
[Nazgul_edited.zip](https://github.com/prusa3d/PrusaSlicer/files/10556934/Nazgul_edited.zip)

This simplified 3MF asserts during elephant foot compensation and crashes during support spots generation. Which of these issues is the one happening in the original 3MF (if any) I don't know.

mcclure3dprints, 2023-02-02:
What if neither elephant foot compensation or supports are turned on, and I still receive this error code? I have a TPU part and ABS part which fit together. Is there a solution to work around this at the time being?

kubispe1, 2023-02-03:
Ok, we fixed it and tested it. In the next release, Access Violation Error should not appear and slicing would be fine.
Closing, Thanks
[image]

closed 2023-02-03 by kubispe1