prusa3d/PrusaSlicer #9490: Text embossing tool - Text align option ?

DaTTcz, 2023-02-01:
The text embossing tool is a great feature !
Is there a setting for text alignment ? (left, center, right) I don't see it anywhere...

Thank you David

kubispe1, 2023-02-01:
At the moment there is no text alignment option.

gudvinr, 2023-02-04:
I hope it gets implemented. Only workaround rn is to manually align text yourself.  
If you add more than 1 line it starts to be quite important.

JosephMarinier, 2026-09-25:
I see text alignment is now in PrusaSlicer 3! 🎉 Along with many other text settings. Just toggle "Advanced" on. Thank you very much!

[image]