prusa3d/PrusaSlicer #9491: Connectors in PrusaSlicer 2.6.0 Alpha 2

pilou33, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
After cutting  a piece with connectors, this connector cannot be inserted easily and I have to machine the connector and the pieces

**Describe the solution you'd like**
When a connect or is generating (great improvement) Prusa slicer could apply it a % of reduction to not have to machine all pieces

lukasmatena, 2023-02-01:
The tolerance is configurable when drilling the holes: 
[image]

pilou33, 2023-02-01:
Hi,

Great I didn't see the option :-), i'll test ASAP
Great job, great improvement

closed 2023-02-01 by pilou33