prusa3d/PrusaSlicer #9492: Post processed gcode not getting uploaded to server

EasyG0ing1, 2023-02-01:
### Description of the bug

According to [this document](https://help.prusa3d.com/article/post-processing-scripts_283913), when post-processing a gcode file that is to be uploaded to a server such as OctoPrint, we are supposed to suffix the filename that is passed to the script with `.output_name`, which I take that to mean that I am to append the filename provided by the `SLIC3R_PP_OUTPUT_NAME` environment variable. So, for example, if:

the gcode file passed into the post-processing code is: `/var/folders/83/2_55ccv57wz23fxkx81556qw0000gn/T/.PrusaSlicer.upload.04bd-86a5-d55a-0fbb`

and the filename provided in `SLIC3R_PP_OUTPUT_NAME` is: `PartFanDuct.gcode`

Then my post-processing code should save the modified gcode to: `/var/folders/83/2_55ccv57wz23fxkx81556qw0000gn/T/.PrusaSlicer.upload.04bd-86a5-d55a-0fbb.PartFanDuct.gcode`

However, when my code does that, the gcode that is uploaded to the server is not the modified gcode that I processed, but rather it is the original Prusa sliced gcode.

I have verified that the post-processed file does indeed exist where it was saved, but that is not the file that gets uploaded to the server.

When I chose the option to post-process to a local file, everything works fine. It is the uploading to the server that isn't working.

Am I misunderstanding the instructions concerning how to do this? If so, what is it that I have incorrect?

### Project file & How to reproduce

[PartFanDuct.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10557153/PartFanDuct.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2

### Operating system

macOS 13.1

### Printer model

Anycubic Mega X

EasyG0ing1, 2023-02-01:
@lukasmatena 

My code is a Java program that takes the path passed into it and loads that file into memory then modifies it.

Where do I save the modified code so that Prusa recognizes it as the correct file to upload to the server?

closed 2023-02-01 by EasyG0ing1

lukasmatena, 2023-02-01:
@EasyG0ing1 Great. As long as you modify the file in place and do not try to rename it, it should be quite simple.