prusa3d/PrusaSlicer #9493: Organic supports ignore the "no supports for bridges" setting

k1-801, 2023-02-01:
### Description of the bug

As the name states, when the type of supports is set to organic, and the setting for "no supports for bridges" is set to "on", the supports keep spawning under every bridge they can find, effectively ignoring the setting completely.

### Project file & How to reproduce

Only tested on this particular model. It has a lot of bridges going on in the middle section, and all of them got a ton of supports underneath, despite the setting for "no supports for bridges" being on.

[image]
[image]

[Ventilation_plate_tree_supports.zip](https://github.com/prusa3d/PrusaSlicer/files/10557384/Ventilation_plate_tree_supports.zip)

On this particular model, not only they are not needed there, but they turned out to be impossible to remove, and the slicer spends a lot of time generating supports- most of this time could be saved by not generating them for bridges.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2+win64

### Operating system

Windows 10 x64

### Printer model

Flying Bear Ghost 5

bubnikv, 2023-02-09:
Fixed (at least partially) with f1977b07be304076be720b058757bc4d36158e83
Organic supports now use the same algorithm as classic supports when blocking bridges.

kubispe1, 2023-02-10:
We checked and tested it. Looks like that fix works and will be part of the next release.
Thanks for your cooperation. Closing (picture is not good, but....)
[image]

closed 2023-02-10 by kubispe1