prusa3d/PrusaSlicer #9494: 2.6.0 Alpha > Importing Anker Profile Error

UncleJessy, 2023-02-01:
### Description of the bug

When using the configuration wizard I'm getting the below error importing the Anker Profile
https://imgur.com/kF9pLPf
[img]https://i.imgur.com/kF9pLPf.jpg[/img]

### Project file & How to reproduce

Open the Configuration Assistant
Select the new Anker Profile
Continue to add
it will show the above error 

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha

### Operating system

Macos 12.1 Monterey M1 Mac

### Printer model

Ankermake M5

kocikdav, 2023-02-01:
Hello. Thank you for reporting.
This is very interesting error since 2.6.0-alpha2 should be first version to introduce Anker profiles. Is it possible that you imported some 3rd party profiles (with Anker printer) into previous version of PrusaSlicer?

kocikdav, 2023-02-01:
I see there is another issue opened (#9497) which I guess happens on the same computer and seems to be related to this one. Would you mind sending us your configuration folder zipped? It would help us greatly finding the issue.

Cisien, 2023-02-01:
If you used the community anker profile in 2.5.0 and imported your config when starting 2.6.0-alpha2, then the community profile is copied over in that process. Adding the anker profile that is included in 2.6.0 will then cause name conflicts and errors. You can go into your configuration folder and delete the AnkerMake.ini/idx/folder files to remove the community profiles and use the shipping profile instead.

UncleJessy, 2023-02-01:
> If you used the community anker profile in 2.5.0 and imported your config when starting 2.6.0-alpha2, then the community profile is copied over in that process. Adding the anker profile that is [...]

Yeah I think that is the issue. I had a the original Anker M5 profile built by trey already installed. Assumed it would be okay since the profiles were not named the same but I guess not. Didnt even think about that

Cisien, 2023-02-01:
I wonder if it makes sense to rename the profile to AnkerMake instead of Anker. I think AnkerMake is technically more correct

just-trey, 2023-02-01:
@UncleJessy I can confirm that if you loaded my profiles manually this indeed would cause this issue. Sorry for the confusion. @Cisien let's discuss naming on discord

kocikdav, 2023-02-02:
@just-trey @Cisien @UncleJessy Would you share the community profile in question to us? I would like to see what exactly caused the problem and if we could prevent it in future.

just-trey, 2023-02-02:
@kocikdav, all versions are here https://github.com/just-trey/AnkerMake-M5-Profile/releases. I was waiting for the release and shared these beforehand. I will probably stop making this public now that it is included, but I need to understand the process of updating better.

closed 2026-09-29 by prusaslicer-bot[bot]