prusa3d/PrusaSlicer #9498: Text embossing tool - wrap around surface

Jakub1177, 2023-02-01:
### Description of the bug

Embossing tool does not wrap the long text around the whole model surface ( cylinder in my case ) and apply the text only on the visible faces by camera.
That means only 180degrees is working and also the text at the edges of visible area is distorted.
This might be a new feature request and not a bug. 
I could not find a release note limitting only visible faces by the camera

### Project file & How to reproduce

[image]
[Shape-Cylinder_embossed.zip](https://github.com/prusa3d/PrusaSlicer/files/10557993/Shape-Cylinder_embossed.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 Alpha2+64bit

### Operating system

Windows 10

### Printer model

MK3S

Jony01, 2023-02-02:
It is about [projection type](https://en.wikipedia.org/wiki/3D_projection). Embossing tool use [Orthographic projection](https://en.wikipedia.org/wiki/3D_projection#Orthographic_projection) (a.k.a. parallel projection) so [cylindric](https://en.wikipedia.org/wiki/Cylindrical_equal-area_projection), [conic](https://en.wikipedia.org/wiki/Map_projection), sphere and curve projection are feature request.

Laserology, 2024-07-20:
Any plans on adding this? Would appreciate it. This would allow adding texture to give parts a nice grip

Jony01, 2024-07-22:
There was add a way around. This feature called "per glyph".
For each letter(glyph) in text is applied [Orthographic projection](https://en.wikipedia.org/wiki/3D_projection#Orthographic_projection) separately with respect to the surface direction under the letter.
[image]

Laserology, 2024-07-22:
> There was add a way around. This feature is called "per glyph".
> For each letter(glyph) in text is applied [Orthographic projection](https://en.wikipedia.org/wiki/3D_projection#Orthographic_projection) separately with respect to the surface direction under the letter.
> [image]
> 

This works good for text, but it wouldn't apply to imported SVG textures, which is mainly what I'm after.

Good to know though! I wouldn't have known that otherwise.

Jony01, 2024-07-22:
Soo ... You could convert your texture pattern into font glyph and than emboss text ;-)

Laserology, 2024-07-22:
> Soo ... You could convert your texture pattern into font glyph and than emboss text ;-)

Actually, that's not a bad idea. 

May consider it, thank you!

closed 2026-09-29 by prusaslicer-bot[bot]