prusa3d/PrusaSlicer #9501: Organic Supports possible new feature or bug

thunderpipesusa, 2023-02-01:
I have been printing 133.4mm tall items with a small single footprint for the last few weeks.  I had to add discs to the bottom (mouse ears) to help stabilize them but they still will wobble slightly on my bed slinger even if I turn down the speed to 25-30 mm/s.  I was really looking forward to the organic supports.  Supports are not needed for the items, I tried painting on supports and using the organics.  I have attempted every change I can under the interface (currently at 0) and with the branch adjustments to get the supports to even touch the model but they wont reach the model at all.  there is a slight gap visible in prusa slicer (no dark green visible) between the model and the tree supports.  I thought maybe it was just a bug, so I loaded to an sd and printed it anyway.  The final result is as was sliced with no contact between the support and the model.  Is there a way to get the tree supports to ignore the support not needed and just contact the model wherever I paint on enforcers?  
[image]
[image]
[image]
[image]

thunderpipesusa, 2023-02-15:
Did this get skipped? Haven’t heard anything on this

Sparhawk402, 2023-02-23:
Is there any update on this? Having the same issue with organic supports not actually touching the model

stepikovo, 2023-03-02:
Hello @thunderpipesusa, please provide us with the project file saved as 3MF. However, there is no "green layer" (the interface) until the angle reaches at least 48° (or 73° for Organic).

thunderpipesusa, 2023-05-18:
looks like the issue is gone with the beta 2.... however these items are hollow inside on purpose, for some reason organic supports are now being printed in the hollow area to meet up with the outside supports.  I will send the 3mf now but am not sure where to send it

thunderpipesusa, 2023-05-18:
Ok i need to zip it.... I am just renaming to .zip not actually compressing fyi
[this copy renamed from 3mf to zip Diamond ORGANIC Supports FINAL - Copy.zip](https://github.com/prusa3d/PrusaSlicer/files/11511928/this.copy.renamed.from.3mf.to.zip.Diamond.ORGANIC.Supports.FINAL.-.Copy.zip)
[this copy was 3mf compressed to zip Diamond ORGANIC Supports FINAL.zip](https://github.com/prusa3d/PrusaSlicer/files/11511930/this.copy.was.3mf.compressed.to.zip.Diamond.ORGANIC.Supports.FINAL.zip)

Dschadu, 2023-09-03:
I'm facing the same issue. Tall print, and I tought the organic supports will be perfect to keep it stable. But they don't touch the model. Now I have a perfectly printed colorfull tree, but a failed model
Using Version 2.6.0+win64

[image]

closed 2026-09-29 by prusaslicer-bot[bot]