prusa3d/PrusaSlicer #9505: Add filament change from CLI

gyscos, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
I would like to slice and generate gcode from a model file using the command-line interface.

**Describe the solution you'd like**
Some CLI option to bring the same features as the GUI:
* specify a height in millimeters where a color change should happen
* optionally specify the new color to use

**Describe alternatives you've considered**
Using custom after-layer gcode, with a condition on the layer height.
Has to be careful to properly account for retraction.

**Additional context**
* The current `--colorprint-heights` appears deprecated and does not seem to be doing much.
* Is it expected that these color changes are not exported as part of the print config?

w5xd, 2024-04-22:
The --colorprint-heights option is documented for the CLI, but does not work. Specifically, the option is parsed and, if syntatically correct it is accepted, but, having traced through a debug build, makes it to the m_config of the Print::ApplyStatus class, where it never gets accessed again. 

I request that --colorprint-heights either be supported (by emitting the --color-change-gcode, which is also apparently documented, but ignored) or its documentation removed. I wasted several hours trying to figure out why the CLI parsed this option but then ignored it.

w5xd, 2024-04-22:
Having looked at this code for the first time less than 24 hours ago, its not obvious from the source why --colorprints-heights was decided to be deprecated, but its clear that there is enough code to support it if the developers wanted to. 

For my own purposes, I have built myself a 2.7.4 PrusaSlicer to run in CLI mode which does apply the --colorprints-heights to the model and emits M600 to the gcode at the appropriate layer like any civilized person would expect. However, I can't tell whether my modifications might meet any other requirements. I attach it as a patch to use at your own risk.

My reasoning, based on source code from tag version_2.7.4: 
In CLI mode, in PrusaSlicer.cpp, CLI::run(), the model(s) is/are read from file using an empty DynamicPrintConfig in the call to read_from_file, but that is without benefit of having applied CLI parameters to a DynamicPrintConfig. The config CLI parameters are ignored because Model::read_from_file is the only code in the archive might apply the colorprints_heights to the model. 

Further down CLI::run, there is this call to Print::apply:<br/>
<code>
                print->apply(model, m_print_config);
</code>

My patch modifies Print::apply() to call update_custom_gcode_per_print_z_from_config
[colorprint-heights.patch](https://github.com/prusa3d/PrusaSlicer/files/15063372/colorprint-heights.patch)

closed 2026-09-29 by prusaslicer-bot[bot]