prusa3d/PrusaSlicer #9506: Account Linking

Boogabaga, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
This is by no means a problem. Prusa Slicer is my primary slicer of choice.

**Describe the solution you'd like**
It would be nice if Prusa Slicer had an account login where your profiles and settings would be available from anywhere you login. 

**Describe how it would work**
I often use Prusa Slicer at home & work, where I have multiple printers at both locations, and being able to have all of my profiles available for all of the printers I use, regardless of where I am, would help to streamline my workflow. Sometimes after I tweak a profile, I forget to export the config. and have to wait till the next day to get it, if I remember. If Prusa Slicer had an account login, I wouldn't have to rely on my terrible memory; the profiles would just be there, as I left them, when I need them.

**Describe alternatives you've considered**
I'm sure you guys are tired of hearing about Bambu Labs, but I like how they've implemented the account feature in their reskin of Slic3r. Regardless of whether I'm at home, or work, when I open Bambu Studio, my profiles, settings, and everything else is just there.  

**Additional context**
Perhaps the only other context would be linking the Prusa Slicer account to the Prusa Research Account and Printables Account, where it would just be one account, versus having multiple accounts for one primary resource.

Thank you for your consideration and I apologize if this is already available and I just missed it.

0x6d64, 2023-02-01:
Hi @Boogabaga, do you have any chance of syncing _files_ between the computers to work on (Dropbox, private repo in GitHub.com)? I'm asking since prusalicer has the ``--datadir ABCD`` command line option where you can point to a directory to use for your profiles. If this directory would be in e.g. your dropbox you could sync across devices with the current version of PS.

Boogabaga, 2023-02-01:
> Hi @Boogabaga, do you have any chance of syncing _files_ between the computers to work on (Dropbox, private repo in GitHub.com)? I'm asking since prusalicer has the `--datadir ABCD` command line [...]

Admittedly, I am not the most savvy when it comes to computers and I'm unsure how to proceed implementing your suggestion. I do not have a Dropbox account, but setting it up through GitHub could be a possibility. Adding accounts to Prusa Slicer with the possibility of a Prusa App for smart phone control of a printer seems like a logical next step for them. Actually, I'm surprised they haven't already implemented it.

0x6d64, 2023-02-01:
Using GitHub would work only in case you are kind of familiar using git to manage files, and also a half-automated process (saving/pushing them into a repository and then downloading/pulling them on the other machine would be something that is triggered by commands entered into the command line or button presses on a git client).

The setup using a service like Dropbox, GoogleDrive or OneDrive would be more automatic. You talked about a work computer, please make sure that you are allowed to use a service like this on your work computer (in a corporate environment this is usually not allowed for security reasons).

Here is a brief description on how you could set this up (paths are examples obviously):

- export you profiles and settings somewhere
- create shortcut to PrusaSlicer.exe and located it e.g. on the desktop
- modify shortcut by right clicking it and select properties, then enter the target as this: ``C:\Users\0x6d64\Downloads\PrusaSlicer-2.6.0-alpha2+win64-202301311700\prusa-slicer.exe --datadir C:\Users\0x6d64\OneDrive\prusaslicer_settings_sync`` (we assume that we use OneDrive, GoogleDrive, Dropbox will have similar folders that they sync)
- start the shortcut, PS will now have no profile or settings initially (we pointed to an empty config dir, thats why), import the exported settings that we created in step 1
- switch to second computer, and check that your sync tool (OneDrive, Dropbox, ....) synced the folder
- create a shortcut, again make it point to the folder with the PS settings
- after opening the shortcut, you now should have automatically the settings from the first computer. Any changes to the setting should result in changed files, those get synced by the file sync tool and then are downloaded on the other computer once it is started

Points to consider:

- this might not work well if you run different versions of prusaslicer
- changing settings on both computers could result in "file conflicts" (since both files are changed, and the file sync tool does not know which file is the right one). Frequent exports into an export are recommended

Boogabaga, 2023-02-02:
I certainly appreciate the time and effort you've put into trying to educate me, and while I certainly have a better understanding of the proposed methodology now, I think I might just migrate everything over to Bambu Studio, since it's basically the same (or not, like I said I am not the most savvy in regards to coding and programming, so please don't take offense) and has the desired feature already implemented. I feel a little more confident in my abilities in going that route. Your points of consideration wouldn't really be of concern, due to the more automated nature of having all the needed information associated to an account.

0x6d64, 2023-02-02:
@Boogabaga, no worries, I don't have any hard feelings. I just wrote down the steps just in case somebody stumbles over this PR, maybe this is helpful to them then. Good luck with your setup, however it may look in the end and happy printing!