prusa3d/PrusaSlicer #9507: Add option to change color at specific place in the layer

cimbalek, 2023-02-01:
**Is your feature request related to a problem? Please describe.**
It could be beneficial to have option to change filament (on single extruder machine) in the middle of the layer (eg. for texts)

**Describe the solution you'd like**
Add plus sign to change filament also to the bottom slider in the layer view

**Describe how it would work**
Clicking plus sighn in horizontal slider would work similarly to the plus sign on vertical slider. Difference would be that you exactly choose the place where to insert M600, contrary to putting into beginning of the layer. I believe this won't need much effort to implement.

**Describe alternatives you've considered**
I have considered manually modifying g-code but it's not really user friendly.

**Additional context**
[image]

rtyr, 2023-02-01:
We already have such request, so I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5195, https://github.com/prusa3d/PrusaSlicer/issues/5951.

closed 2023-02-01 by rtyr

cimbalek, 2023-02-01:
Sorry. I searched if it was here before creating a ticket, but there is so many issues I haven't found it.