prusa3d/PrusaSlicer #9508: Less ringing by corner de/acceleration

Mike8-I, 2023-02-01:
This is a suggestion for improved print quality.
I first suggested this in a discussion about motion compensation in the firmware section.
Alessandro Pantaleon from Prusa support suggested, that I offer the idea here. I did a search if something similar already exists, but did not find anything. Maybe I used the wrong keywords for the search......
So this is it:
"I would like to offer an idea that should give better looking prints without loosing (maybe even gaining) print speed:
The root cause for the ringing are resonances that are excited by harsh accelerations or jerks ( visible on the outer perimeter).
Instead of compensating the motions of the resonance (fixing symptoms) we could avoid to excite the resonances (fixing closer to the root cause).
This can be done in the slicer. Before reaching a corner, the slicer decelerates with an adjustable rate and then also accelerates with this same rate after the corner. This could be tested with only the outer perimeter or all perimeters. The ( adjustable ) corner-acceleration that works could be somewhere in the range 10 to 100mm/s x s.
As this does not influence infill, it would not do to much damage regarding print time. I could even be possible, that faster prints are possible, because the perimeter speed can be chosen higher. The ringing would not be affected by the higher speed, as the corner acelleration would still be small, just the acceleration distance would be longer."

DragonFinlay, 2023-02-03:
Basically [Pressure Equalizer](https://help.prusa3d.com/article/pressure-equalizer-_331504), but for every corner?
How would you define such a corner, what's the angle threshold? What about very complex models?

Mike8-I, 2023-02-03:
Thanks for the question, I thought a little bit more about that.
We also need a corner speed. Definition: a straight line is 0°. The corner speed is the perimeter speed multiplied by the absolute
value of the  cosine of the corner angle. Examples: For 20mm/s perimeter speed that results in the following corner speeds: 10° - 19.7mm/s. 45°- 14.1mm/ s  90° - 0mm/s (the movement stops into the corner and accelerates out of it) 135°- 14.1mm/s. Into and out of the corner the corner acceleration is used.
This should do the job for any model complexity, as it accounts for the amount of resonance excitation of the affected axis according to the angle of the corner, whilst giving the fastest possible speed.

Mike8-I, 2023-02-08:
Interesting input, thanks. I feel that nozzle diameter is not relevant, just the arcs ( excitation of resonant systems). I may be wrong, that needs testing. 
The example you show suggests the following strategy: if a sharper corner further up requests a slower corner speed at the current (slight) corner, than the sharper corner "wins".
I'm also thinking about further versions of the slicer where G2/ G3 commands ( radiuses) are used. I will give input there, when I have thought it through.
Linear advance will help to make good extrusions. Regarding pressure equalizer: we might need a switch which one precedes. For a Bowden system this might be pressure equalizer, for a direct drive extruder ( even with flex material) this might be corner decel.
I am aware that this feature needs experiments.  My primary goal is that the MK3, even without resonance compensation, can produce top notch print quality with fast print speeds.

Mike8-I, 2023-04-06:
With the actual slicer releases, the acceleration for the outer perimeter can be set separately. This solves the issue. When setting the acceleration for the outer perimeter low (10 to 7.5mm per square second)  I experienced prints with low to no visible ringing. The perimeter speed can be chosen higher, so long and straight segments will print fast.

closed 2023-04-06 by Mike8-I

polyphia, 2023-11-29:
This does not fix the issue, setting external perimeters alone doesn't give enough granular control over corners and other sudden changes of direction that will cause increased vibrations . 

The op solution would allow to maintain high printing speed of external perimeters while reducing accelerations on non linear movements to reduce ghosting effectively . 

Reducing external perimeters speed as a whole has a very noticeable effect on print speeds, especially in vase mode . 

For reference, using an acceleration of 10mm/s in a small ringing tower model doubles printing time from 35 to 70 mins . 

Using an external perimeter acceleration of 1000mm/s (of a printer with 5000mm/s default) adds just 2 mins to total printing time but has a negligible effect on ghosting .