prusa3d/PrusaSlicer #9511: Alpha2 crashes on models with infill modifiers

mix579, 2023-02-01:
### Description of the bug

For some—but not all—models with modifiers I get the Alpha-2 to crash when I select Slice now. I have only tested it with models that had infill modifiers. But then again, some models do not crash. 

### Project file & How to reproduce

Here's one that immediately crashes when Slice now is selected
[Dremel Holder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10561599/Dremel.Holder.3mf.zip)

It's a simple infill modifier.  
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6-alpha 2

### Operating system

Macbook Air M1 running MacOS Ventura 13.2

### Printer model

Mini+ and Mk3S+

lukasmatena, 2023-02-01:
Thanks, reproduced. We will fix it. This crash leads to `Layer::sort_perimeters_into_islands`.

bubnikv, 2023-02-02:
Fixed with 3cb2f5f58f952bc104260c84292411894f957552

kubispe1, 2023-02-03:
As @bubnikv issue with modificators and support was fixed and tested.
I am closing
[image]

closed 2023-02-03 by kubispe1