prusa3d/PrusaSlicer #9512: Organic supports that land on the print do not print a brim and pop off easily.

Cisien, 2023-02-01:
### Description of the bug

[image]

The organic supports that land on the build plate do get a brim and have had no problems with sticking to the plate.

### Project file & How to reproduce

Slice anything where organic supports will land on the build plate. For the example above, i'm printing the gridfinity drawer holder with automatic supports:

https://thangs.com/designer/Machenna56k/3d-model/Gridfinity%20Drawers-188528

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2

### Operating system

Windows 11

### Printer model

Ankermake M5

kubispe1, 2023-02-01:
Thanks for the issue. We already noticed this problem. Hope it will be fixed soon. See you.

kubispe1, 2023-05-17:
In PS 2.6.0-beta1: 
The bottom interfaces of Organic supports have been fixed and enabled, providing enhanced stability when positioned on the model's body.
I am closing the issue. 
Thanks

EddiOS42, 2023-07-21:
Just spoke with customer support with this issue on release version 2.6.0. Confirmed the bug still persistent.

PekoKent, 2024-05-01:
Same issue with 2.7.4, still not fixed. Setting "Bottom contact Z distance" to 0 does not work, nor does 0.1
Found a simple workaround: just set it to 0.01     ;-)

ProMate04, 2024-06-04:
Can confirm this problem is still, present in the latest version, 2.7.4. I saw that there were no interface layers generated at the bottom, and knew there would be problems with my print. I tried it anyways, and miserably failed.

PeoKent's workaround somewhat works, what it does is it prints the support tree directly on top of the model. This still does not generate interface layers, but should work for some scenarios. It is hacky, and it would be amazing if this actually got fixed.

How it looks:
[image]

How it looks with PekoKent's suggestion (0.01 bottom contact Z distance):
[image]

KelanMorgan, 2024-07-31:
While this fixed the problem temporarily, it would be nice to have the ability to use interface layers instead