prusa3d/PrusaSlicer #9514: Organic supports overlapping into the model

steven509, 2023-02-01:
### Description of the bug

When there are multiple models with organic supports some of the organic supports overlap into the actual model.

[image]

[image]

[image]

[image]

### Project file & How to reproduce

[organic-supports-overlap.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10562733/organic-supports-overlap.3mf.zip)

The layout of the items in the example are a bit nonsensical but it shows the issue.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2

### Operating system

macOS 13.2

### Printer model

MK3S+

mfraser, 2023-02-02:
I'm also seeing this.
[image]
[image]
[image]
[Webcam V1 Bracket.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10567137/Webcam.V1.Bracket.3mf.zip)

foreachthing, 2023-02-02:
You need to import your model multiple times (not making instances) and then select all and MERGE.
Then the support will see the objects as one and will no longer interfiere.

kubispe1, 2023-02-02:
I am closing as a known issue and duplicate problem... thanks @foreachthing  for an explanation of the current solution.

closed 2023-02-02 by kubispe1

kubispe1, 2023-02-13:
There is reference #316