prusa3d/PrusaSlicer #9515: 2.6.0-alpha2 - PressureEqualizer: Error parsing a float

sarusani, 2023-02-01:
### Description of the bug

I get the error "PressureEqualizer: Error parsing a float" when I try to slice any model with an overhang. This happens as soon as I use pressure equalizer, no matter what value I set it to.

### Project file & How to reproduce

Open attached project file, slice.
[x-carriage-back.zip](https://github.com/prusa3d/PrusaSlicer/files/10562903/x-carriage-back.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha2+win64

### Operating system

Windows 10 64bit

### Printer model

MK3S+

kubispe1, 2023-02-02:
I can confirm that this issue is reproducible. This combination makes us trouble. Thanks #9485

kubispe1, 2023-02-09:
Thank you for your cooperation issue was tested. The fix will be part of the next release.

closed 2023-02-09 by kubispe1