prusa3d/PrusaSlicer #9517: Organic supports extending through model

StevenWeathers, 2023-02-02:
### Description of the bug

Organic Supports are extending through the model instead of just supporting the areas marked for support (or auto generated).

[image]

### Project file & How to reproduce

[fide-knight-organicsupport.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10564468/fide-knight-organicsupport.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2+MacOS-x64

### Operating system

macOS ventura 13.2

### Printer model

Prusa MK3S+

wilecoyote2015, 2023-02-02:
I can reproduce this on Arch Linux with 2.6.0-alpha2

neophyl, 2023-02-02:
Again can you please read the release notes.  
The following -
The Organic supports currently pose the following limitations:

For large models with a lot of supports, 16GB RAM minimum is recommended.
Top / bottom vertical separation distances are rounded to multiples of layer height.
**Variable layer height is not supported.**
Interface layers are not always produced reliably. As of now printing with zero interface layers is recommended, which produces easy to remove supports.
Rafts with Organic supports are buggy.

Your project contains variable layer height.

MrBlackmidi, 2023-02-02:
Are there any plans to eliminate the limitations of organic supports for the layer releases of 2.6?

lukasmatena, 2023-02-02:
@MrBlackmidi Yes.

kubispe1, 2023-02-02:
Closed as duplicate issue - #9462

closed 2023-02-02 by kubispe1

MrBlackmidi, 2023-06-25:
hey guys, are there still plans to eliminate these limitations because i noticed that these limitations were not mentioned in the ps 2.6 release notes on the prusa website?

MrBlackmidi, 2023-12-12:
Hi guys, is there an update on the removal of the limitations for organic supports?