prusa3d/PrusaSlicer #9518: Program crashes, error dialog reads The instruction at 0x00007FFD681A8AB0 referenced memory at 0xFFFFFFFFFFFFFFFF. The memory could not be read.

SeanI-Git, 2023-02-02:
### Description of the bug

Using PrusaSlicer 2.6.0 Alpha2, and when I slice the model the program crashes, displaying a dialog box that reads "The instruction at 0x00007FFD681A8AB0 referenced memory at 0xFFFFFFFFFFFFFFFF. The memory could not be read." The model has organic supports and infill modifiers. It is relatively simple but contains 24 instances.

I expected it to slice the model successfully.

[image]

### Project file & How to reproduce

[Project with PS 2.6.0 Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/10564785/Project.with.PS.2.6.0.Issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha2+win64

### Operating system

Edition	Windows 10 Home Version	21H2 Installed on	‎6/‎11/‎2022 OS build	19044.2486 Experience	Windows Feature Experience Pack 120.2212.4190.0

### Printer model

Prusa MK3S+ MMU2S

kubispe1, 2023-02-02:
Organic supports + modifiers - this could be fixed with this issue #9511

kubispe1, 2023-02-03:
issue with modificators and supports was fixed and tested.
I am closing this issue
[image]

closed 2023-02-03 by kubispe1