prusa3d/PrusaSlicer #9519: Maximum height reached, but model does not reach maximum height

powpingdone, 2023-02-02:
### Description of the bug

[image]

When slicing this model at a 45deg angle (with some variablility +-5deg), the maximum height error is emitted despite the model being significantly less than the maximum Z height (210, while the max height is 160). 

### Project file & How to reproduce

Project file: [broken slicing.zip](https://github.com/prusa3d/PrusaSlicer/files/10566320/broken.slicing.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha2-linux-x64-GTK3

### Operating system

Fedora Silverblue 37 inside a toolbox (essentially fedora 37)

### Printer model

MK3S

DragonFinlay, 2023-02-03:
When exporting the model as STL and re-importing it, the issue is gone. Same if I then save the 3MF and re-open it, no issue slicing it. Not sure why, we'll try to find out more about it.

atten-dev, 2023-03-01:
I also encountered this on PrusaSlicer-2.6.0-alpha4+win64-202302180012 and did a few experiments I hope will be helpful.

In this zip file is the STL, a 3MF where it slices OK (Good.3mf), and a 3MF where it shows the height error (Bad.3mf). The only difference is the object has a 45 degree rotation around the object's Z axis.
[MaxHeightExceeded.zip](https://github.com/prusa3d/PrusaSlicer/files/10858342/MaxHeightExceeded.zip)

To reproduce without the 3MF file:
1. Set printer to be a Prusa MINI
3. Click the +Box icon on the top bar (any other method of adding the STL should be OK too)
4. Select the object
5. With world coordinates mode selected:
6. Rotate around X axis by 55 degrees
7. Rotate around Z axis by 45 degrees
8. The object should now slice OK
9. Select Object coordinates mode
10. Rotate around object Z axis by 45 degrees
11. Should now give error that print is taller than maximum allowed height.

Replacing 45 degrees in step 9 by any multiple of 45 degrees also shows the issue. Replacing it with 90 degrees or any multiple will not show the max height issue. Quite interesting.

Alternatively, you can reproduce the issue with only world coordinate rotations:
1. Rotate around Z by 45
2. Rotate around X by 55
3. Rotate around Z by 45

Note: I've also maybe possibly observed some non-determinism in this. One time I ran through steps 1-7 and it immediately didn't work without doing steps 8-10, but wasn't able reproduce the issue without steps 8-10 again. Or maybe I didn't do my steps right...

If I scale the object down to 83.6%, it will slice. (83.7% will not slice.) This is roughly (but not exactly!) 150mm in height.

config.ini: [config_ini.txt](https://github.com/prusa3d/PrusaSlicer/files/10858471/config_ini.txt)
System info:
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 19045), 64-bit edition
Blacklisted libraries loaded into PrusaSlicer process:
C:\ProgramData\A-Volute\A-Volute.28054DF1F58B4\Modules\ScheduledModules\x64\NahimicOSD.dll

kubispe1, 2023-05-17:
This issue should be resolved in regards to 2.6.0 beta1. 
If you encounter any problems, please let us know. Closing this issue. Thank you.