prusa3d/PrusaSlicer #9521: Cannot move Object on print plate

joernupbde, 2023-02-02:
### Description of the bug

When I place any object onto the print plate in the slicer. I can no longer move it in X, Y or Z direction. It keeps snapping back into the center.

### Project file & How to reproduce

[Form-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/10567113/Form-Box.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha2+win64-202301311700

### Operating system

Windows 10

### Printer model

Prusa MK3S+

lukasmatena, 2023-02-02:
I'm guessing you have enabled "Auto center parts" in Preferences.

joernupbde, 2023-02-02:
@lukasmatena Your guess is correct. Sorry for bringing this up wrongly. Thanks for your help.

closed 2023-02-02 by joernupbde