prusa3d/PrusaSlicer #9522: [PS 2.6.0 A2] crashes when generating gcode for sequential printing

Ohmarinus, 2023-02-02:
### Description of the bug

Sequential printing turned on with standard settings.

3 small objects spaced out across the bed.

Crashes when generating gcode (at 90%).

**Doesn't crash when generating gcode for the same object with `sequential printing` turned off.**

### Project file & How to reproduce

[sequential_crash_PS2.6.0_A2.zip](https://github.com/prusa3d/PrusaSlicer/files/10567244/sequential_crash_PS2.6.0_A2.zip)

I turned on `sequential printing` with the standard settings.

Pressed `generate gcode` and then it goes to 90% and after that it crashes PS to exit. It's a rather simple error.

I also noticed that the third object that I copied using the `add instance +` button has a wrong outline:
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

 PrusaSlicer-2.6.0-alpha2+win64-202301311700_signed.zip 

### Operating system

Windows 10

### Printer model

DIY Makerbot Replicator '3x' running RRF with a Fly CDY v3 board

lukasmatena, 2023-02-02:
Thanks, reproduced. I checked the backtrace and it is the same crash as #9464.

Ohmarinus, 2023-02-02:
> Thanks, reproduced. I checked the backtrace and it is the same crash as #9464.

Thanks for clarifying, I only search for the term 'sequential' before submitting the bug, it appears you're right and these crashes are related.

lukasmatena, 2023-02-03:
Fixed in [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3). Closing.

closed 2023-02-03 by lukasmatena