prusa3d/PrusaSlicer #9523: Separate temperature for bridges in filament profile

k1-801, 2023-02-02:
**Is your feature request related to a problem? Please describe.**
ABS is known for difficult bridging. To achieve good bridges on it, one has to drop the temperature to 230 degrees or even lower, but when the entire model is printed with this temperature, it will not have any layer adhesion, so the part will crumble from very little force. Printing bridges at high speeds usually helps to lay out the correct geometry in layers above it, even on high temperatures, but the strings of the bridge will remain separated from the model and from each other. But when the bridge is short, the printer simply doesn't have enough time to reach the required speed, and the bridge sags.

**Describe the solution you'd like**
An additional setting for printing bridges at a different temperature than the rest of the model could solve this without having to compromise between structural strength and bridges.

**Describe how it would work**
When the printer is about to start a bridge, instead of doing it right away, instead, it should go to a homing position, turn on the part fan to the bridging level and wait for the temperature to drop to the expected value, and only then proceed to print th bridges of this layer. After the bridges are done, it should do the same to re-heat to the regular printing temperature.
The overhanging perimeters on the sides of this bridge might become a problem or a weak spot, since they would have to either be printed at low temperature together with the rest of the perimeter, making the entire perimeter weak, or be printed at the regular high temperature, potentially allowing sagging. There is an option to try and print the "cold" segments of the perimeters first and then continue them with hot plastic.
The downsides would be slightly longer print time (~40 seconds per layer with bridges), although no real life print would have bridges or overhaging perimeters on every layer, so it should be fine.
I am currently testing it by manually placing the g-code for this exact trick before and after the bridging layer, though, it is really inconvenient to do so, and the g-code markers (as well as color change markers) keep disappearing for no apparent reason.

k1-801, 2023-02-03:
Additionaly, the part fan on some printers may take up to 1 second to get to speed, which is enough for the printer to lay down several short bridges/bridging perimeters, which won't get any proper cooling. Allowing the printer to wait up to a second to let the fan get to speed could also help.

audkar, 2026-09-26:
Still relevant