prusa3d/PrusaSlicer #9524: incorrect organic supports

ivosak, 2023-02-02:
### Description of the bug

incorrect organic support location once cutted with connectors

### Project file & How to reproduce

don't know, but enclosing 3mf file
[General Pavel flannel.zip](https://github.com/prusa3d/PrusaSlicer/files/10567711/General.Pavel.flannel.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-Alpha2

### Operating system

WIN 10

### Printer model

Prusa Mini+

ivosak, 2023-02-02:
It looks it is caused by Variable layer height

kubispe1, 2023-02-02:
@neophyl thanks. You are helping us quite lot. Cudos for you. 
I am closing this issue as duplicate #9462

closed 2023-02-02 by kubispe1