prusa3d/PrusaSlicer #9526: Organic Support Raft

AlexSilva0992, 2023-02-02:
### Description of the bug

Not possible to use raft with organic supports

### Project file & How to reproduce

Slice

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.6.0-alpha2

### Operating system

Windows 11

### Printer model

Prusa MK3S+

neophyl, 2023-02-02:
From the release notes-
The Organic supports currently pose the following limitations:

For large models with a lot of supports, 16GB RAM minimum is recommended.
Top / bottom vertical separation distances are rounded to multiples of layer height.
Variable layer height is not supported.
Interface layers are not always produced reliably. As of now printing with zero interface layers is recommended, which produces easy to remove supports.
**Rafts with Organic supports are buggy.**

kubispe1, 2023-03-01:
Hi, thanks for the issue. In the next release, there should work raft OR brim properly in combination with organic supports. If you find some complications after release 2.6.0-alfa5 will be out let us know. Also, see our changelog that will come with.
Closing the issue.

closed 2023-03-01 by kubispe1