prusa3d/PrusaSlicer #9528: Support interface interferes with organic supports

cmerkle, 2023-02-02:
### Description of the bug

When generating organic support with support interface layers, the support interface seems to overlap with the organic supports, rather than printing on top of them. This will cause nozzle drag on the top layers of the organic supports, and likely will cause the organic supports to be knocked over. 

### Project file & How to reproduce

[Ender_3_v2_tool_holder.zip](https://github.com/prusa3d/PrusaSlicer/files/10569099/Ender_3_v2_tool_holder.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha2+win64

### Operating system

Windows 10

### Printer model

Ender 3 V2

SockedIn, 2023-02-02:
I had this issue with some organic supports indeed being knocked off.  I've done one print with interface set to zero as suggested in the release notes.  The print completed normally.  I'll have to test further.

neophyl, 2023-02-02:
Please read the release notes.  Specifically this section Ive marked in bold.

The Organic supports currently pose the following limitations:

For large models with a lot of supports, 16GB RAM minimum is recommended.
Top / bottom vertical separation distances are rounded to multiples of layer height.
Variable layer height is not supported.
**Interface layers are not always produced reliably. As of now printing with zero interface layers is recommended, which produces easy to remove supports.**
Rafts with Organic supports are buggy.

commit 5b94971f (2023-02-03): Organic supports: Added check for variable layer height, with which
Organic supports are not compatible.
Fixes #9528 and similar.
Moved the object max Z against build volume Z to Print::validate().
Updated layer_height_profile_from_ranges() to compress the resulting
ranges.
closed 2023-02-03 by bubnikv