prusa3d/PrusaSlicer #9530: Crash with complex model (2.6.0 alpha 2)

the-ress, 2023-02-02:
### Description of the bug

PS crashed after adding a complex model. I had PS set to default settings with 0.2mm/Prusament PETG/MK3S+ and added the model. Then I switched to another app. When I wanted to switch back to PS, it wasn't there (and I didn't saw any error window either), but a crash dump file got created.

I'm running PS with `--sw-renderer` (because reasons).

The model is https://cults3d.com/en/3d-model/game/albedo-genshin-impact (I can send the STL privately if needed.)

### Project file & How to reproduce

Crash dump: [prusa-slicer.exe.15080.zip](https://github.com/prusa3d/PrusaSlicer/files/10570004/prusa-slicer.exe.15080.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha2+win64

### Operating system

Windows 11 (22621.1194)

### Printer model

Prusa MK3S+

lukasmatena, 2023-02-03:
@the-ress Thanks. The dump leads to an event handler reacting to a keypress, which tried to open some tool from the left toolbar. The crash happened when undo/redo tried to capture the snapshot. It is difficult to tell more without full memory dump, but it may be enough. I will look through the relevant code later.

[image]

the-ress, 2023-02-03:
Thanks for the details. I'll try to reproduce it and capture a full dump.

the-ress, 2023-02-05:
Good news! I managed to reliably reproduce it:
1. Open a new instance
2. Press Ctrl+I
3. Paste STL path (preferably a complex one, but it should work with any STL if you're fast enough in step 5)
4. Press Enter
5. Rapidly press L for a few seconds

Here's the full memory dump: https://tomcova-my.sharepoint.com/:u:/g/personal/tereza_tomcova_onmicrosoft_com/ESptOVzFmHBIgOB2Fs7h32oBRsRcgcqeZ7CONbtlBcWuKw?e=S4xIA4
(There should be a download button in top left corner.)

closed 2026-09-29 by prusaslicer-bot[bot]