prusa3d/PrusaSlicer #9532: Draft Shield not disabling

inventorAdam5, 2023-02-02:
### Description of the bug

When draft shield is set to disabled, and skirt height is set to >1, a draft shield is still generated at skirt height.
[image]
[image]

### Project file & How to reproduce

[Hole Test v1.zip](https://github.com/prusa3d/PrusaSlicer/files/10570852/Hole.Test.v1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha2+win64-202301311700

### Operating system

Windows 10

### Printer model

Anycubic Kobra

neophyl, 2023-02-02:
You have skirt height set to 15.

kubispe1, 2023-02-02:
@neophyl thanks for the explanation. I am closing this issue. The skirt is applyed...

closed 2023-02-02 by kubispe1